jnthas/clockwise #7: E Pin on wiring diagram is incorrect

abrender, 2022-09-26:
The image at [https://github.com/jnthas/clockwise/blob/gh-pages/static/images/display_esp32_wiring_bb.png](https://github.com/jnthas/clockwise/blob/gh-pages/static/images/display_esp32_wiring_bb.png) shows the E PIN connected to GPIO32 on the ESP32, however the code in firmware/src/main.cpp has `mxconfig.gpio.e = 18;`.

This might lead some folks to wire up the display with an incorrect E PIN.   Since GPIO18 is used on the ESP32 Trinity, maybe it makes the most sense to update the diagram to show GPIO18 instead of GPIO32?

jnthas, 2022-09-28:
Nice catch! I think I have the Fritizing file on my laptop, gonna check this weekend. Thank you.

jnthas, 2022-10-03:
Updated pin E wiring in the image. Thanks again.

closed 2022-10-03 by jnthas