jnthas/clockwise #10: Add Pacman clock like mario-bros-master

nodoubtman, 2022-12-21:
Hello..

Can you add Pacman, as mario-bros-master

i don't know how to assemble all this...

Thanks.
Marc.

Lofty73, 2022-12-21:
Pac man is included in the web based installer follow the instructions on the main clockwise page

nodoubtman, 2022-12-21:
I know :)
But i want to customize it, like to display the date in french, etc :)

Thank you so much!!!: )
Marc.

Lofty73, 2022-12-21:
Apologies I understand

jnthas, 2022-12-21:
Hi @nodoubtman, in order to have the Pacman set please clone the repo https://github.com/jnthas/clockwise first. Then change directory to `clockwise/firmware/lib` and clone the Pacman's clockface https://github.com/jnthas/cw-cf-0x05 there. You should be good to go.

nodoubtman, 2022-12-21:
Thank you so much :)
Working :)

closed 2022-12-21 by nodoubtman

nodoubtman, 2023-02-01:
> Hi @nodoubtman, in order to have the Pacman set please clone the repo https://github.com/jnthas/clockwise first. Then change directory to `clockwise/firmware/lib` and clone the Pacman's clockface https://github.com/jnthas/cw-cf-0x05 there. You should be good to go.

Excuse me my ignorance, but how do i open it with platform io?

Thanks.
Marc.