jnthas/clockwise #11: Doesn't seem to be working for P3 matrix

zombodotcom, 2023-01-09:
[image]

This turns on the board and displays correctly during setup.

With 
```
HUB75_I2S_CFG::i2s_pins _pins={R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN};
HUB75_I2S_CFG mxconfig(
                      64,   // width
                      64,   // height
                      2,   // chain length
                   _pins
);
```

this setup also works for SmartMatrix and ESP32-HUB75-MatrixPanel-I2S-DMA

it's connected and saying collision detected but nothing is being drawn. I am using the mario sketch. and included the clockface.h and stuff from that folder. 

[image]

I have also gotten the wifi to work. Nothing works past the initial test that the panel Pinout is working.

zombodotcom, 2023-01-09:
this seems to have to do with the [ESP32-HUB75-MatrixPanel-DMA](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA) library. it will display RGB for 

```

  dma_display->fillScreenRGB888(128,0,0);
  delay(1000);
  dma_display->fillScreenRGB888(0,0,128);
  delay(1000);
  dma_display->clearScreen();  
  delay(1000);  
  ```

at start. but the fastLED side isn't working, so far. 

The plasma pattern works, but its not calling anything in fastled other than color.

closed 2023-01-09 by zombodotcom

zombodotcom, 2023-01-31:
for anyone else having this issue,

you need to edit the 
`ESP32-HUB75-MatrixPanel-I2S-DMA.h` file to be the correct MATRIX_HEIGHT and MATRIX_WIDTH

[image]

also 

the default pins

[image]

jnthas, 2023-02-02:
I'm happy you found the solution and share it here. Thanks