jnthas/clockwise #13: Improv Wi-Fi

balloob, 2023-03-09:
ESP Web Tools has built-in support for [Improv Wi-Fi over Serial](https://www.improv-wifi.com). You can see it in the video here: https://youtu.be/E8bdATqXM8c?t=50 (this is the old UI, it has a network selector now).

Have you considered adding support for Improv Wi-Fi?

jnthas, 2023-03-09:
Look pretty interesting. I'm considering now add support to it. Any help is appreciated :)

jnthas, 2023-03-13:
@balloob I missed a simple example on how to use the Improv Wi-fi. I started working on that and there is chance to create an Arduino library to handle to protocol using the SDK provided. I would like to see more people adopting Improv Wifi, it's a cool feature

https://jnthas.github.io/improv-wifi/
(ignore my code right now, I'm just testing :D)

balloob, 2023-03-13:
That's awesome!

We never got to creating the Arduino example because we don't use that with ESPHome.

jnthas, 2023-04-03:
@balloob  A new version of Clockwise was released today now with support to Improv WiFi!

Check the demo video here:
https://twitter.com/jnthas/status/1642691071246229504?s=20

closed 2023-04-03 by jnthas

zombodotcom, 2023-05-04:
this has removed the AP functionality where if it now is required to setup wifi over the webpage. there is no AP to connect to if it cannot connect to WIFI.