jnthas/clockwise #16: Build fails with ESP-IDF 4.4.4 because of missing fil

derryx, 2023-04-06:
Hi!

Building with ESP-IDF 4.4.4 fails. There seems to be some files missing in the tree:
C:/Development/ESP32/clockwise/firmware/lib/cw-commons/WiFiController.h:3:10: fatal error: ImprovWiFiLibrary.h: No such file or directory
 #include "ImprovWiFiLibrary.h"
          ^~~~~~~~~~~~~~~~~~~~~

I am using the main branch.

jnthas, 2023-04-06:
Hi @derryx. Thanks for noticing that. I just updated the submodules files. could you try to build it again please?

hallard, 2023-06-07:
Not sure it's relevant, but I'm experiencing same issue here fresh IDF 4.4 installed from VSCode, 
Fresh clone with submodules of main branch and build try

```

idf_main.dir/main.cpp.obj -MF esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj -c /Users/charles/Devt/clockwise/main/main.cpp
In file included from /Users/charles/Devt/clockwise/main/../firmware/src/main.cpp:7,
                 from /Users/charles/Devt/clockwise/main/main.cpp:5:
/Users/charles/Devt/clockwise/firmware/lib/cw-commons/WiFiController.h:3:10: fatal error: ImprovWiFiLibrary.h: No such file or directory
 #include "ImprovWiFiLibrary.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
```

abrender, 2023-06-11:
@derryx I have a bunch of changes that should fix this issue and enable the project to be compiled with the `ESP-IDF` platform.

jnthas, 2023-06-11:
Thank you @abrender. I was trying to fix that issue with ImprovWiFiLibrary yesterday. You opened the first PR to your fork, I don't know if it was a mistake but I already applied the change you did there.

jnthas, 2023-06-11:
Guys, it should be working now. Thanks @abrender again. I'm preparing a new release for today.

closed 2023-06-11 by jnthas

abrender, 2023-06-11:
@jnthas I will add a github action/workflow that will automagically build the project with `esp-idf` on each pull request so that should hopefully help prevent breakages

jnthas, 2023-06-11:
That's perfect

hallard, 2023-06-12:
Just to let you know that now it's building fine, thanks for the fix.