jnthas/clockwise #17: defined night time switched off during night time

burkadius, 2023-04-07:
Hello Jonathas, thank you very much for the new SW 1.1.0. I have installed it and everything works perfectly. Especially good for me is the regulation of the display brightness because at night in the dark the display brightness is disturbing. 
It would be very practical for me if the display could be switched off completely at a defined time of night. Alternatively, an LDR ABC (automatic brightness control) would also help.
Best regards
Burkhard

jnthas, 2023-05-13:
Hey folks. I'm working on it. I need a little help, I would like to know the min value of bright that display don't turn off for you. For me, it's 4, when I decrease it more, the display turn off.

burkadius, 2023-05-14:
Hi, yes, value 4 will be ok. 
Tanks for help.
Best regards
Burkhard

Puncher4444, 2023-05-14:
For me, 3 is the minimum value of brightnes so the display wont turn off - everything below 3, the display stays black.

jnthas, 2023-05-14:
Ok, thanks guys. I just release a new version with automatic bright control. Let me know what do you think. Don't forget to set the values in the settings. More details in README.md

burkadius, 2023-05-23:
Hi Jonathas, finally I had time to test the new release. I have installed an LDR and the ABC controls the display. All is well. But in complete darkness the display does not switch off completely. If I set Automatic Bright to 0:0 (ABC = OFF) and Display Bright to 2, the display stays OFF, at 3 it is ON but very dim. Is it possible to change the SW so that at ABC 0:800 the display is completely OFF and only changes to ON at >0, e.g. 1:800?
Best regards
Burkhard

jnthas, 2023-06-04:
Hi @burkadius. I got your point, it is quite easy to do that change. Thanks for the idea, I have not thought this. The change was done, it will be available in the next release.

jnthas, 2023-06-11:
Done in this release -> https://github.com/jnthas/clockwise/wiki/Connecting-the-LDR

https://twitter.com/jnthas/status/1667988156220620801?s=20

closed 2023-06-11 by jnthas

burkadius, 2023-06-15:
Hello Jonthas
I have installed the new version. Everything is great, works really well for me now. The display is off at night and comes on at dawn in the morning.
Thanks a lot
Burkhard

Von: Jonathas Barbosa
Gesendet: Sunday, 11 June 2023 22:16
An: jnthas/clockwise
Cc: burkadius; Mention
Betreff: Re: [jnthas/clockwise] defined night time switched off during nighttime (Issue #17)

Closed #17 as completed.