jnthas/clockwise #18: API endpoint to change brightness / turn on/off the screen.

vandalon, 2023-04-15:
Hi, Thanks for this amazing peace of ready made software!

I'm wondering if it would be possible to create and API endpoint to set the brightness and/or turn on/off the display for use with home automation. I would like to hang a couple of these clocks in the kids bedrooms but it would be nice if they would dim in the evening and turn off when it's time to sleep :)

Thanks again for all the effort already put in this project!

jnthas, 2023-04-15:
Hi @vandalon. Since version 1.1.0 it's available those endpoints. Let me know if it works for your case.

Set display bright (0-255):
```
curl -X POST 192.168.18.36/set?displayBright=8
```
(at least for a while) You have to restart the device to have the change completed 
```
curl -X POST 192.168.18.36/restart
```

vandalon, 2023-04-15:
That works, thanks! 
Small remark. the restart command will cause curl to timeout (of a timeout is set) because the system instantly restart without returning a 200 or 204 (as the set command does) :)

vandalon, 2023-04-15:
Also, short question, should the system reboot if wifi is disconnected longer then a certain period of time? I tried connecting to the device after I did some work on my wifi and it was unreachable until I restarted it. It did still run though.

jnthas, 2023-04-15:
I noticed the same thing, I should create a flag and finish the request before restart. Will fix in the next release. Thanks. 
No, it tries connecting for some time and than it stops, you should see a red wifi sign on it.

vandalon, 2023-04-15:
Yes, this is why I ask, when wifi fails on boot, it will stay there forever then, It might be an idea to have it try a reboot every 10 minutes or so?

jnthas, 2023-04-15:
I don't know... I don't know if it's worth restarting when password is wrong or network is unreachable.

vandalon, 2023-04-15:
Well, I was thinking from a kids perspective. If something went wrong with the wifi it would be nice if it recovered itself, of course if it's just a faulty password it won't fix anything, but if the wifi was offline for some reason it would help. Also I noticed that sometimes when I reboot my esp it will just fail connecting to wifi, until I reset it again.

jnthas, 2023-04-15:
Yes, it happens with me too, for any reason it simple can't connect on network and I have to restart it. I don't know why yet. I think I can add a parameter to decide if it will restart or not if connection fails.

vandalon, 2023-04-15:
That would be awesome :)
I have another, unrelated, question: The line in the world map clock face, what does it represent? :)

jnthas, 2023-04-15:
Good question. If I remember well, it's the Noon of each part of the world. If you use this clockface let me know if the line is over your timezone at 12:00

vandalon, 2023-04-15:
Well if I change my timezone the line shifts. But it should still be noon in the same place, regardless of my timezone ;)

jnthas, 2023-04-15:
Well, for me it's noon in the middle of pacific haha. The red line is close to New Zealand, and it's 9:47 there. Looks good :)

jnthas, 2023-05-14:
Hi, I made some fixes in this latest version including the restart after some time offline and return HTTP response before restarting.
https://github.com/jnthas/clockwise/releases/tag/v1.2.1
Closing the issue

closed 2023-05-14 by jnthas