jnthas/clockwise #20: Ruins the AP functionality

zombodotcom, 2023-05-04:
with the new Improv library, it no longer can be setup standalone without wifi credentials. No AP is emitted if it cannot connect to wifi and needs to be setup on the computer.

zombodotcom, 2023-05-05:
after upgrading to show someone to show it was a self sufficient  product. you legit ruined my day. the device deserves wifi to setup apart from the now defacto only way " Connect ur shit  by usb and set ur wifi in the improv prompt" 

guess what improv prompt doesnt always work to let you set ur wifi after the flash. this is blasphemous to its original inception that, any user could setup wifi without a computer, just connect to the clockwise AP if it cant connect to WIFI. input credentials/

I will make my own way. but holy shit, you took someones word for a library, and you forgot what you made it for......

abrender, 2023-05-05:
> after upgrading to show someone to show it was a self sufficient product. you legit ruined my day. the device deserves wifi to setup apart from the now defacto only way " Connect ur shit by usb and set ur wifi in the improv prompt"

> guess what improv prompt doesnt always work to let you set ur wifi after the flash. this is blasphemous to its original inception that, any user could setup wifi without a computer, just connect to the clockwise AP if it cant connect to WIFI. input credentials/

> I will make my own way. but holy shit, you took someones word for a library, and you forgot what you made it for......

@zombodotcom - you are commenting on an open source project whose author has posted their work for free. If you are not happy with a decision that the owner of the project has made then you have multiple options:

1) Roll back to a revision before the offending change was made.
2) Create a pull request with a proposed solution.
3) Fork the repo and create your own version
4) Create an issue *politely* describing the problem and engage the owner & community in an effort to change course.

closed 2023-05-05 by zombodotcom

zombodotcom, 2023-05-05:
done