jnthas/clockwise #25: Connecting LDR

anthonyjclarke, 2023-05-16:
Thanks for the great addition, can you please confirm how to connect the LDR to Pin35 for your code to operate?

Is it as per this ESP Trinity schematic? 
[image]

https://github.com/witnessmenow/ESP32-Trinity/blob/master/hardware/ESP32Trinity/V1.2/Schematic_ESP32-Trinity_2021-08-27.png

jnthas, 2023-05-17:
Hi Anthony. Sorry, it's exactly this. I'm going to update the documentation.

TheGroundZero, 2023-05-28:
If you get the Trinity, is it a matter of just soldering the LDR on the provisioned pads?

Lofty73, 2023-05-28:
Yes just solder one to the marked position.

hallard, 2023-06-02:
Awesome software, would be great to be able to select GPIO used for LDR in settings page to avoid users to recompile.
I'm Author of an old OpenSource [shield](https://github.com/hallard/WeMos-Matrix-Shield-DMA) for ESP32 D1 Mini and as I wanted to keep compatibility with ESP8266 D1 Mini, LDR is connected to A0 AKA `GPIO36` on ESP32 D1 Mini

jnthas, 2023-06-03:
@anthonyjclarke I added the LDR wiring in the [README](https://github.com/jnthas/clockwise#connecting-the-ldr) 
@hallard Thanks for the feedback, I will put this change in the backlog. If you don't mind, I will link your shield as one of the options to run the Clockwise.

closed 2023-06-03 by jnthas

anthonyjclarke, 2023-06-04:
> Awesome software, would be great to be able to select GPIO used for LDR in settings page to avoid users to recompile.
>
> I'm Author of an old OpenSource [...]

+1 to this. I have some of Charles great boards as well as the I2S board from @witnessmenow boards and would like this flexibility

jnthas, 2023-06-04:
I agree. I will include the option to configure the GPIO in the next release.

hallard, 2023-06-04:
> I agree. I will include the option to configure the GPIO in the next release.

Awesome, thanks 🙏

hallard, 2023-06-04:
> 

> If you don't mind, I will link your shield as one of the options to run the Clockwise.

Of course you’re welcome