jnthas/clockwise #26: New version 17/05

Lofty73, 2023-05-17:
I am encountering problems still with the Access point on the clocks. I finally got the Pac Man clock with the shortened days but now the web interface seems broken. I download the file and eventually get it to accept my wifi password but it say connected but the display does not reset. I cannot get the web page to display and when i reset my screen it seems to log onto my network and hang on NTP server.
I have tried different downloads and two different ESP devices.

jnthas, 2023-06-05:
Hi Nigel, I merged a pull request right now that can be useful for you. It will be possible to change the NTP server in the settings page, I did some tests this weekend and there are some servers way faster than the default we are using. For example, the google one. I let you know when I release the version, probably at the end of the week. 
https://gist.github.com/mutin-sa/eea1c396b1e610a2da1e5550d94b0453

jnthas, 2023-06-11:
Nigel, I released a version with NTP Server. Please, try to change and let me know if it works
https://twitter.com/jnthas/status/1667988156220620801?s=20

Lofty73, 2023-06-11:
Thanks I will in the morning and let you know. Much appreciated

closed 2023-10-08 by jnthas