jnthas/clockwise #27: Configure Wi-Fi Scanning for networks never completed

tonyjobson, 2023-05-23:
Hi, I just took a fresh ESP32 Trinity out of the box. The flash tool found it and paired to it and installed the firmware but the Scan button never finds any wifi networks.
I can connect to the serial and reset the device and see the device complain at lack of wifi and timezone settings.

Unsure if i got a bad device or this is a firmware bug. Can i run an older version?

Can i set the WIFI AP and Password via the console?

I tried Mario clock and PacMan face so far. same behaviour and i've flashed each a few times.

tried two USB cables in case it was a power issue on different USB ports on my laptop.

=========================
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
E (401) esp_core_dump_flash: No core dff�partition found!
E (401) esp_core_dump_flash: No core dump partition found!
[  1023][E][Preferences.cpp:483] getString(): nvs_get_str len fail: timeZone NOT_FOUND
[  1024][E][Preferences.cpp:483] getString(): nvs_get_str len fail: wifiSsid NOT_FOUND
[  1028][E][Preferences.cpp:483] getString(): nvs_get_str len fail: wifiPwd NOT_FOUND
[  2168][E][Preferences.cpp:483] getString(): nvs_get_str len fail: timeZone NOT_FOUND
[  2168][E][Preferences.cpp:483] getString(): nvs_get_str len fail: wifiSsid NOT_FOUND
[  2172][E][Preferences.cpp:483] getString(): nvs_get_str len fail: wifiPwd NOT_FOUND
============================

tonyjobson, 2023-05-23:
OK.  I managed to get it past this by powering the trinity with an external power supply. looks like the board may have been very powe hungry...

jnthas, 2023-05-23:
That's right. You need an external power supply of 3 amps or more. It's not the board, but the led matrix that consumes all that power.

closed 2023-05-23 by jnthas