jnthas/clockwise #29: Feature Request: Allow setting custom NTP servers

TheGroundZero, 2023-05-31:
I prefer to keep IoT devices on a non-internet VLAN where possible.
I also run my own NTP server on the local network and communicate its address via DHCP to clients.

It would be nice to be able to configure my own NTP server either in the Settings page or have the device follow the DHCP options for NTP servers.

TheGroundZero, 2023-05-31:
I'm working on this in a fork, but I'm not used to working with ArduinoIDE.
So I'm looking to incorporate this: https://github.com/ropg/ezTime#setserver-and-setinterval

TheGroundZero, 2023-05-31:
Can anyone try to compile this and check if it works?
https://github.com/TheGroundZero/clockwise/commit/8a1e3f3c9b47f95e3bde1d3aa69961591d40ffe9

jnthas, 2023-06-03:
@TheGroundZero it was a great first attempt. It almost work at first. I made some changes in your fork I would like to push the changes and then you open a PR

TheGroundZero, 2023-06-05:
@jnthas I've initiated the PR and marked it as editable by the maintainer

PR #31 (2023-06-05): Add custom NTP setting
Fixes #29
closed 2023-06-05 by jnthas