jnthas/clockwise #32: Turn the display on/off (or adjust brightness) remotely

joelhauxwell, 2023-06-06:
I built this project last week for my 6yo son and he LOVES it - thankyou for an amazing project.

It would be awesome if it could be controlled remotely - maybe via MQTT?  He isn't in his room most of the day (while as school!) and i could set up an automation in home assistant/node red to turn it off/on.

Unfortunately I do not have the programming skills to add this feature and have no idea how big the task would be to know if its even reasonable to ask!

Some pics of the build for anyone interested (i used a photo frame from IKEA and 3D printed a bezel):

[image]

[image]

[image]

[image]

jnthas, 2023-06-09:
That's amazing! I don't know if you like, but a matte acrylic in front of display with the frame will be even better. Congrats anyways. Please, if you share the design of 3d printed parts I will find some place to make it available for everyone.

Answering your question, it's possible make an HTTP request and set the display bright to zero. I replied a similar question here https://github.com/jnthas/clockwise/issues/18

joelhauxwell, 2023-06-14:
Thankyou for the link to the previous question/answer about remote control - i have been able to integrate it with Home Assistant and its working great!

Of course id be happy to share more details of what I put together - just let me know when there is a place.

Joel.

closed 2023-06-14 by joelhauxwell

mateuszdrab, 2023-06-29:
Wow, I like that frame @joelhauxwell. 
I've just put my display on a tablet stand as I wasn't sure what to use as a frame.

I guess the best place to share would be your own github repo with a readme page and perhaps the 3d design files, maybe the link to the Ikea product. 

If you do this, I'll definitely make one for myself.

jnthas, 2023-06-30:
I agree. Please, share it in a Github repo or on websites like Thingiverse.