jnthas/clockwise #35: Allow `espressif/esp-idf-ci-action` action

abrender, 2023-06-11:
Hi @jnthas,

I'd like to use the [`esp-idf-ci-action` action](https://github.com/espressif/esp-idf-ci-action) for integration tests when new PRs come in for the project.

If you'd like to allow this to work, please use the following steps:
(1) Go to the Repo "Settings" page
(2) Click Actions -> General
(3) Select "Allow jnthas, and select non-jnthas, actions and reusable workflows"
(4) Add this entry for "Allow specified actions and reusable workflows": `espressif/esp-idf-ci-action@*`

Thanks!

jnthas, 2023-06-11:
Done!

abrender, 2023-06-11:
@jnthas thanks! Oh, it looks like `actions/checkout@*` needs to also be added: https://github.com/jnthas/clockwise/actions/runs/5236901026

Thanks!

jnthas, 2023-06-11:
Updated! Can you test again?

abrender, 2023-06-11:
Looks good - https://github.com/jnthas/clockwise/actions/runs/5236994454/jobs/9454888208 - thanks!

closed 2023-06-11 by abrender