jnthas/clockwise #39: Timezone broken?

TheGroundZero, 2023-06-12:
Now with 1.3.0 installed, my Clockwise still doesn't show the correct time.
At first I blamed it being on a non-internet network, as it did work on my 1 installation on an internet-connected network.
However, it still appears to be _stuck_ on UTC, thus being 2 hours off.

I tried configuring my own NTP server as well as reverting back to pool.ntp.org and opening up the firewall for NTP requests comming from my Clockwise.

[image]

JeffWDH, 2023-06-12:
@TheGroundZero it's not obvious in Clockwise's documentation but it uses the eztime library which needs port 2342 (UDP) open to determine your timezone.

https://github.com/ropg/ezTime 

"timezoned.rop.nl is ezTime's own timezone service that it connects to. It is a simple UDP service that gets a packet on UDP port 2342 with the request, and responds with a packet that holds the POSIX information for that timezone (after OK ) or the error (after ERR ). It will only respond to the same IP-number once every three seconds to prevent being used in DDoS attacks."

jnthas, 2023-06-13:
Thanks @JeffWDH  for the clarification. I believe @Lofty73 is passing for this issue. 
You can test it running `nc -u timezoned.rop.nl 2342`, type the time a zone like "Europe/Brussels" and then hit Ctrl-D.

Lofty73, 2023-06-13:
Sorry for the delay in this I will get my displays on the bench and download today. I cannot understand that the tetras clock programmed by @witnessmenow is rock solid at getting the time. I have no problem getting the Internet but the displays currently hang on getting the time.

TheGroundZero, 2023-06-14:
Thank you for researching this.
Ideally, blocking 3 would not _break_ the flow. This shouldn't be a _requirement_.
If possible, I'd like to keep these devices offline.

TheGroundZero, 2023-06-16:
@JeffWDH Would you know if it excpects anything specifically from the EZTime server?
Or could I use a redirect on my router to forward it to a random service (maybe even my actual NTP server) so its connection doesn't fail?

JeffWDH, 2023-06-17:
jnthas provided this information above but essentially it maps a timezone to a POSIX timezone string.

Example:
echo -n "America/New_York" | nc -w 1 -u timezoned.rop.nl 2342
OK America/New_York EST5EDT,M3.2.0,M11.1.0

You could run your own server by grabbing the code here:
https://github.com/ropg/ezTime/tree/master/server

You could then use NAT, redirect the DNS or even modify the code to point to your own server. 

Alternatively (this is untested), if you only care about your own timezone you could run something like this on your own network which would output the string:
socat -U UDP-LISTEN:2342,fork SYSTEM:'sleep 1; echo -n "OK America/New_York EST5EDT,M3.2.0,M11.1.0"'

This is pretty dumb and just waits a second and echoes that string whenever a new connection is made so I'm not sure if it will work.

TheGroundZero, 2023-06-20:
Will look into turning that into a podman container so I can run it next to Home Assistant.
That should do the trick for now.

TheGroundZero, 2023-06-30:
Got something put together but doesn't seem to fully function yet
https://github.com/TheGroundZero/ezTimeContainer/pkgs/container/eztime-server

TheGroundZero, 2023-07-03:
Looks like I got it to work :)
https://github.com/TheGroundZero/ezTimeContainer/releases/tag/v1.0

I also added a DNS override that points `timezoned.rop.nl` to my Podman host and set a NAT rule to redirect `udp/2342` to the same system.

Will monitor a few days to check if it actually stays online.

JeffWDH, 2023-07-19:
@TheGroundZero would you be interested in testing a patch? I've added a manual Posix string option and it works for me. When set you no longer need the eztime server to set a timezone.

https://github.com/JeffWDH/clockwise/tree/manual_posix

https://github.com/JeffWDH/clockwise/commit/360b579d94a2f30c59c055fe834602abb5827d39

JeffWDH, 2023-07-19:
[image]

TheGroundZero, 2023-07-20:
> @TheGroundZero would you be interested in testing a patch? I've added a manual Posix string option and it works for me. When set you no longer need the eztime server to set a timezone.
> 
> https://github.com/JeffWDH/clockwise/tree/manual_posix
> 
> [JeffWDH@360b579](https://github.com/JeffWDH/clockwise/commit/360b579d94a2f30c59c055fe834602abb5827d39)

Sure thing

But you'll have to run me through doing this, I'm afraid.

Lofty73, 2023-07-20:
Watching with interest. New pc built great speed but clocks still are an hour behind. Missed the posix timestring posts so will be interesting to try.

JeffWDH, 2023-07-20:
I hacked together a copy of the web flasher here:
https://www.jhdc.ca/posixtest/

Only the Pacman clockface is available. If this test build works for you guys I can submit a PR to hopefully get this added to the project.

JeffWDH, 2023-07-20:
A list of the Posix timezone strings can be found here:
https://github.com/JeffWDH/clockwise/blob/manual_posix/PosixTimezoneStrings.txt

TheGroundZero, 2023-07-21:
Flashing went smooth and the device booted with immediately the correct time 👍

jnthas, 2023-07-21:
Hey folks, that's an excellent job! This parameter will help many other users I think. I've tested here and it worked great. Merging on master right now.

jnthas, 2023-07-21:
@Lofty73 could you test it flashing from the URL @JeffWDH provided -> https://www.jhdc.ca/posixtest/ 
For London the posix timezone is -> `Europe/London GMT0BST,M3.5.0/1,M10.5.0`
This string means 
> Summer time begins on March’s third Sunday at 1:00am, and ends on October's fifth Sunday at 0am.

closed 2023-10-08 by jnthas