jnthas/clockwise #40: API endpoints

mateuszdrab, 2023-06-29:


jnthas, 2023-06-30:
Hello Mateus. I didn't have time to create a documentation for that yet (it's a good idea), but you can take a look at the code. The handling of path and methods is very rudimentary but you can get the idea there https://github.com/jnthas/clockwise/blob/9e3924bcb3825f85069d636e52a583853bb443f7/firmware/lib/cw-commons/CWWebServer.h#L80C6-L80C6

And the path `/set` can be used with the following parameters
https://github.com/jnthas/clockwise/blob/9e3924bcb3825f85069d636e52a583853bb443f7/firmware/lib/cw-commons/CWPreferences.h#L9

mateuszdrab, 2023-07-02:
Thanks Jonathas

I was just having a look to see what I can do. I also wanted to make sure that WiFi password can't be read from the device, in case it goes into AP mode.

Is it possible to disable AP mode somehow?

jnthas, 2023-07-04:
There is not an easy way to do it, you would have to recompile after removing this line 
https://github.com/jnthas/clockwise/blob/cb6291876f8637978725eea3f3f071c1c9888162/firmware/lib/cw-commons/WiFiController.h#L98C24-L98C24

I didn't understand your question about the Wifi Password, as far as I know, the wifi password can't be read in the WiFi Manager as I don't set it during the AP setup.

PR #47 (2023-09-22): Add api documentation
This PR adds a separate API documentation site that should give an overview of the currently offered API calls. For easier testing and access an OpenAPI specification file (openapi.yml) is being provided that can be used with your favorite tools to call the API.

Fixes #40
closed 2023-10-08 by jnthas