jnthas/clockwise #43: automatic brightness  not stable

heidepiek, 2023-07-23:
Finally got this beautiful clock running. But run into another little problem. Got an ldr connected to gpio 34. In the clockwise settings changed the pin to 34. But whatever value I put in the settings page at automatic bright. The brightness keeps switching back and forth when I turn off the lights. The display changes brightness every couple off seconds. What am I doing wrong? Maybe a wrong ldr? Have tried several types.  But get stuck every time.

jnthas, 2023-07-24:
Hi! This is probably happening because the value read by the LDR is oscillating and it's enough to change the bright of display.  Have you tried to read the value when the room is dark and light? This is my suggestion, with the room dark, click on the "Read Pin" and get the value, do the same with the light on. Put the values in the automatic bright settings page. If it doesn't work, I think using a masking tape to block a little bit of the light can avoid this oscillation. 

Reference here
https://github.com/jnthas/clockwise/wiki/Connecting-the-LDR

[image]

heidepiek, 2023-07-24:
I`m sorry jonathas, this dummy had not seen that there was also a wiki. Had already googled on the internet how to connect an ldr to an esp. But that was already in your wiki. Tried what you said, the values at dark were around 1100 (though my computer screen was on) And at lights on 4095. If I enter these values, it is still not right. But fill in the value 0 and 800, it does work satisfactorily. So how it works exactly I don't know, but it works now. Of course there is also a difference in the ldr`s. I have a few types lying around. Must have something to do with the resistance. Beautifully made jonathas! Nice of you to share that with fellow tinkerers.
Again sorry, reading comprehension is also a profession!  

Greetzz from holland
Harrie

jnthas, 2023-07-25:
Anytime, Harrie! Don't worry, the wiki is a little bit hidden indeed :)

closed 2023-07-25 by jnthas