jnthas/clockwise #44: [Feature Request] Web firmware updates

JeffWDH, 2023-07-23:
I think most people use this project for a clock that sits on a shelf, wall, or other location that may not be near a computer. It would be handy to have the ability to upload new firmware to the device without physically connecting to it.

Fortunately, this is pretty easy on an ESP32 using Arduino:
https://github.com/espressif/arduino-esp32/blob/master/libraries/ArduinoOTA/examples/OTAWebUpdater/OTAWebUpdater.ino

A prerequisite for this IMO would be to implement some sort of authentication on the web interface. We wouldn't want any unauthenticated user to have access to change settings / potentially upload malicious code.

@jnthas what are your thoughts on this?

jnthas, 2023-07-24:
I agree. Update it remotely was one of the things I thought but in that time I used the PxMatrix to drive the display and it frozen when I updated via OTA, then I gave a break for this feature. 
Nowadays, my plan is to use the same Improv protocol I use today for Serial but for [BLE](https://www.improv-wifi.com/ble/). The problem is, I have to implement it from scratch because there is no library for that. I did it for [Serial](https://github.com/jnthas/Improv-WiFi-Library) and I would like to do the same for BLE.

robegamesios, 2023-09-19:
Yes, having the ability to update the firmware without having it connected to your computer would be awesome. how would you integrate https://github.com/espressif/arduino-esp32/blob/master/libraries/ArduinoOTA/examples/OTAWebUpdater/OTAWebUpdater.ino to the project? thanks

jnthas, 2023-09-22:
The project already have a webserver, I believe it's not that hard to implement this feature, add a couple more endpoints and process like you guys sent as example. I'm creating a tag for this.

robegamesios, 2023-10-08:
@jnthas I was able to do a sample of Mario clock and I was able to upload a bin file to upload it. not clean but worked for me. see attached. 
1. upload the otaDefaultUpdate to esp32.
2. then access the ip address and upload the Mario clock bin file
3. I added a reset wifi button. 
4. Setting wifi is the same in wifi manager. then include the timezone 

Like you said, if it can get implemented as an endpoint to upload the bin file that would be super awesome. thanks

[mariobros-clock.zip](https://github.com/jnthas/clockwise/files/12840005/mariobros-clock.zip)
[otaDefault
[mariobros-clock.ino.bin.zip](https://github.com/jnthas/clockwise/files/12840012/mariobros-clock.ino.bin.zip)
Update.ino.bin.zip](https://github.com/jnthas/clockwise/files/12840011/otaDefaultUpdate.ino.bin.zip)
[otaDefaultUpdate.ino.zip](https://github.com/jnthas/clockwise/files/12840013/otaDefaultUpdate.ino.zip)

jnthas, 2023-10-08:
Hi Rob. Thank you! Could you open a PR so we can merge it or work on a final version together.

robegamesios, 2023-11-12:
@jnthas @JeffWDH check this out.

1. Ive included all the current clockwise themes and Audio Visualizer which can be selected and the led matrix panel will update to it and restart itself.  
2. i also added OTA.
3. I'm having issues with the Canvas clockface though, if the audio visualizer is included i get a No memory error when deserializing the json. maybe you can figure it out? I also tried to make the Audio visualizer a clockface but i still get the no memory error.
4. right now the only way the only for me to use the canvas theme is to upload firmware via OTA. would be great to not have to upload it as a firmware. 

Still prototyping, can use more improvement. 

https://github.com/robegamesios/clockwise/tree/feature/add_audio_visualizer

[image]
[image]