jnthas/clockwise #45: [Feature Request]

FabianWeyerstall, 2023-07-26:
Is it possible to use this LDR Sensor with your Out of the Box Clockwise version? 

https://de.aliexpress.com/item/32648002929.html?spm=a2g0o.productlist.main.91.3bb35cbbJCel8r&algo_pvid=7b60ab7a-91fb-4bee-8bce-148f8f87c61c&algo_exp_id=7b60ab7a-91fb-4bee-8bce-148f8f87c61c-45&pdp_npi=3%40dis%21EUR%210.40%210.35%21%21%210.43%21%21%402145280e16903917914547086d0762%2110000001111815755%21sea%21DE%212103824747&curPageLogUid=P2nasSHnZjtv

Thanks in advance.

Ps.: Best Clock I ever had :)

jnthas, 2023-10-08:
Hi Fabian, thank you. It is possible but you can simplify it using only the LDR. If you are using a Trinity board, there is placeholder in the board or if you are using  a ESP32 devboard, connect it like in the diagram here -> https://github.com/jnthas/clockwise/wiki/Connecting-the-LDR

hoho1986, 2024-11-07:
I cannot find the resistor value of A9050. What is the recommended value?

jnthas, 2024-11-17:
The A9050 is an Light Dependent Resistor or LDR

jnthas, 2025-10-14:
LDR will vary its resistence value depending on the light. I'm closing this issue, feel free to reopen later

closed 2025-10-14 by jnthas