jnthas/clockwise #48: Feature request more 64x32 options

Steph1116, 2023-09-25:
Hi.

I am very new to esp32 boards. I've been using matrix 64x32 and 64x64 boards for a year or so with raspberry pi pico. But I'm completely dumb as anything and have a really really hard time getting a simple project to work😪. I found your project and wow it's so simple to plug, install and play. Downside to clockwise atm is it only has one (Tetris) 64x32 option. Can anyone please add to the 64x32 library please please please. A simple clock, and/or clock & date possibly with a brightness slider and colour picker. 

If not I completely understand. But thanks anyhow 😊. 

Side note, my 64x64 matrix panel doesn't have a D pin, just labeled GND, so I can't get it to work. I followed the pin out and I get the image but,(if split in 4 top to bottom) the top and 3rd only showing the image. 

I just have the esp32, jumper wires and the matrix, nothing else and I can't buy other hardware for a while as I've spent way too much already. TMI, but in case it was going to be suggested. 

Thanks!!!!

Steph1116, 2023-09-26:
Hi, here is a few boards. There are 34 chips, let me know which ones you'd need. [image][image][image][image]

board707, 2023-09-26:
SM5166 is a miltiplexor of the standard "three_to_eight" type, the Sm16208 is a LED driver.
As a conclusion, this is a normal panel, that should work fine with the library.

Steph1116, 2023-09-26:
But yet it doesn't for some reason. I hand wired it with jumper wires one at a time, I also used the 16 wire ribbon the panel came with. Is it possible it's the esp32? [image][image]

Steph1116, 2023-09-26:
Fully appreciate your attempt. I'll upload a photo of what the panel looks like after I wire it up again and install a clock.

Steph1116, 2023-09-26:
I got the 64x64 working. My jumper wires and ribbon that came with the panel seemed to be the culprit. I found some jumpers I bought a while back, opened the package, wired both up and bingo! Very sorry to trouble you all with what apparently was a non-issue. But I'm still looking for projects for the 64x32 that are plug and play like the Tetris clock.

jnthas, 2023-09-26:
Hey guys! @Steph1116 I think your problem is not far to be solved. I have been working in a clockface where you can draw the clock the way you want just using this editor https://editor.clockwise.page/
For a while it will work for 64x64 because this value is hardcoded in the firmware, but is something easy to change [here](https://github.com/jnthas/clockwise/blob/3c8b6016d5f577161c47d9c008ff268c004552a1/firmware/src/main.cpp#L30C1-L31C1) 
Please, if you want, try to setup the dev environment (you will have to recompile the project) and I can give you some support

Steph1116, 2023-09-26:
Hey @jnthas Thanks for replying. My problem with that is I'm far from understanding all of that. At the moment I understand plug and play. I'm learning as I go, but I'm far from throwing in anything close to looking like a clock for quite some time.  My end goal is to have a clock that is white during the day and red at night (10pm-6am). It would be date (Sept 26 '23) and time (12-hour) and maybe a third row for weather, conditions and temp. But as mentioned, I'm far from creating something like that myself. I truly truly appreciate all the hard work everyone puts into these project, and i wish one day I could contribute to it. Thanks!!!

Steph1116, 2023-09-26:
Oh, and with the clocks that are currently on clockwise, and the Tetris clock, do they have the option to turn off the led on the esp32 itself?

closed 2023-09-26 by Steph1116

jnthas, 2023-09-27:
Ok @Steph1116, no problem! Keep visiting us because I think it's something possible to do it and maybe someone can help with this. There is no parameter to turnoff the builtin led for now.

Steph1116, 2023-09-27:
Thanks! Do you take care of both the 64x64 and 64x32 projects?

jnthas, 2023-09-27:
Nope. @witnessmenow that developed the Tetris clock

Steph1116, 2023-09-27:
Awesome, I'll contact him. Do you know how to turn off the onboard LED's on the esp32 through clockwise?