jnthas/clockwise #49: Time in words clock

Steph1116, 2023-10-02:
Happened to be awake at midnight to notice that the clock says noon at midnight lol

jnthas, 2023-10-08:
Oh, probably the timezone was incorrect. It  was displaying the time AM where it should be PM. 
When is 0 hour, it shows midnight, take a look:
https://github.com/jnthas/cw-cf-0x02/blob/18d4cd9a08146c1725bee974c7b01e070ea6df01/DateI18nEN.h#L49C1-L50C1

closed 2023-10-08 by jnthas

Steph1116, 2023-10-08:
Thanks for looking into it. Do I need to do anything to fix it or is it auto updated? No big deal either way as it wasn't a bothersome thing.

jnthas, 2023-10-08:
Did you set your timezone on the settings page? If it's correct, it should do it automatically. Also, try to check the 'Use 24h format' also in the settings page

Steph1116, 2023-10-08:
The timezone was set correctly upon first install to show the correct time. I use 12-hour format as I'm not interested in the 24hour clock. I'll go into the settings and just restart the clock and I'll see tonight at midnight if it is displayed correctly and report back if it isn't. Thanks

Steph1116, 2023-10-11:
So it's been a few days, first couple of days I forgot and fell asleep but was able to see tonight that it's still showing noon. I check all my settings and everything is set correct. So was the code corrected? Should I format and reinstall the clock? Or does it need to be first? Just for the record, I know how to install it and understand a bit of the code, but I am still in kindergarten when it comes to my level in understanding all of it. Thanks!