jnthas/clockwise #51: [Feature Request] Integration with Home Assistant

jorgeassuncao, 2023-10-03:
Is it possible to integrate the Clock with Home Assistant? Being based on ESPhome i believe it would be realy simple to do. 

At the moment when trying to integrate it with HA this is the result

[image]

jnthas, 2023-10-08:
Hi Jorge. This is an interesting feature that I didn't work yet. I agree that it won't be difficult to add it. For now, I think you can use the endpoints available on Clockwise to do a basic integration with Home Assistant 
https://clockwise.page/docs/api

karl0ss, 2024-01-02:
I created 2 `rest_command` in my HA yaml 

```
rest_command:
  clockwise_set_display_brightness:
    url: "http://<CLOCKIP>/set?displayBright={{ brightness }}"
    method: POST
  clockwise_restart:
    url: "http://<CLOCKIP>/restart"
    method: POST
    content_type: "application/x-www-form-urlencoded"
```

You can then execute the services using 
```
service: rest_command.clockwise_set_display_brightness
data:
  brightness: 25
```

```
service: rest_command.clockwise_restart
```

I have then created automations tied into "day and night routines" that will execute the first rest command, setting the brightness to X, sleeping for 2 seconds, then executing the restart service,

closed 2025-10-14 by jnthas