jnthas/clockwise #53: how to convert png to base64 string?

robegamesios, 2023-10-10:
hi, what tool do you use to convert the png to base64? i tried different online converters but i get this error when running.

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
E (456) esp_core_dump_flash: No core d�f.�partition found!
E (457) esp_core_dump_flash: No core dump partition found!
[WiFi] Connected to ASUS-RT-AX56U-2.4G, IP address 192.168.50.112
[Time] NTP Server: time.google.com, Timezone: America/Sao_Paulo
[HTTP] GET request to '192.168.50.10/peach-gomo1.json' on port 4443
deserializeJson() failed: InvalidInput

thanks

jnthas, 2023-10-11:
Hi Rob! I've implemented a button in the right side of the image that converts it. Just pay attention to the image size, it must be smaller than the display resolution, in this case 64x64 pixels

[image]

robegamesios, 2023-10-11:
i noticed that if the png file is less than 1kb, then it works, otherwise only a black screen is shown.

robegamesios, 2023-10-11:
> Hi Rob! I've implemented a button in the right side of the image that converts it. Just pay attention to the image size, it must be smaller than the display resolution, in this case 64x64 pixels
> 
> [image]

nice! what is that app?

jnthas, 2023-10-11:
It's a work in progress, but it's functional

https://editor.clockwise.page

robegamesios, 2023-10-11:
> https://editor.clockwise.page

thanks let me try it now!

robegamesios, 2023-10-11:
i still see a black screen, the size is 64x64

[image]

jnthas, 2023-10-11:
Export the PNG removing all metadata so the size will be smaller 
I recorded a video creating a theme, I think I export the PNG there: 
https://www.youtube.com/watch?v=8PtY-WK-Q-g

Clock Club is the place where you get the examples 
https://github.com/jnthas/clock-club

robegamesios, 2023-10-11:
> Export the PNG removing all metadata so the size will be smaller I recorded a video creating a theme, I think I export the PNG there: https://www.youtube.com/watch?v=8PtY-WK-Q-g
> 
> Clock Club is the place where you get the examples https://github.com/jnthas/clock-club

ok thanks, let me look at those. 

btw the 24h format does not work for me, even though i unchecked it, it still shows 24 hour format

jnthas, 2023-10-11:
it might be a bug.. are you using the mario clockface?

robegamesios, 2023-10-11:
> it might be a bug.. are you using the mario clockface?

No im using the canvas.  the mario clock face works fine.  

Do you have the .ino file for the other clocks like pacman and canvas? so i can try to add the OTA code and test them.

jnthas, 2023-10-11:
Oh ok. For Canvas I'm using the ezTime to format the the date/time so you have all the flexibility
https://github.com/ropg/ezTime#getting-date-and-time
Capital H is for 24h, lower h is 12h.

jnthas, 2023-10-11:
You have to set up your environment using PlatformIO. I didn't use Arduino IDE, so I don't have the .ino. 
Follow the instructions here -> https://github.com/jnthas/clockwise#how-to-change-the-clockface-platformio

robegamesios, 2023-10-11:
> Oh ok. For Canvas I'm using the ezTime to format the the date/time so you have all the flexibility https://github.com/ropg/ezTime#getting-date-and-time Capital H is for 24h, lower h is 12h.

ok, the endpoint from the clockwise website does this POST:

http://192.168.xx.xx/set?use24hFormat=0

the values are 0 and 1 but doesn't affect the result. still shows only 24h format

jnthas, 2023-10-11:
Yes, because on Canvas you configure it on the JSON. Just inform the desired format here https://github.com/jnthas/clock-club/blob/e3f6f9105a045a62db7abc55a34319182061f0da/shared/donkey-kong.json#L19C1-L20C1

robegamesios, 2023-10-11:
@jnthas i tried even gimp, where did you get your assets which are less than 1kb? i think that is my main issue, i could not find any assets that are less than 1kb even after i removed all the extra info.  i was able to find a random png less than 1kb and that worked. thanks again

robegamesios, 2023-10-11:
@jnthas  one thing im curious is the green background im getting, its suppose to be transparent background but i get green color. when you get a chance, can you try the json file below and see if you get the same thing? thanks

zip:
[peach-gomo-1.zip](https://github.com/jnthas/clockwise/files/12875136/peach-gomo-1.zip)

json:
[peach-gomo-1.json](https://github.com/jnthas/clockwise/files/12875124/peach-gomo-1.json)

robegamesios, 2023-10-12:
@jnthas i also noticed this issue, sometimes it does not flicker but sometimes or most of the time it does. happens right after i pushed the json and restart. any thoughts? thanks

https://github.com/jnthas/clockwise/assets/10041871/d32d3a94-0ed3-425b-bc44-980698484413

robegamesios, 2023-10-12:
also what is the range for setting the background color? i tried setting this android purple color but nothing shows up 11691775 (#b266ff)?

jnthas, 2023-10-12:
Hello Rob! I can test it tonight. Pretty nice the theme you are working, let's make it work. The flicker is new to me, but I can take a look too, maybe the delay or the image size. Let's see.

robegamesios, 2023-10-12:
> Hello Rob! I can test it tonight. Pretty nice the theme you are working, let's make it work. The flicker is new to me, but I can take a look too, maybe the delay or the image size. Let's see.

thanks @jnthas . here are some more json files that flickers most of the time.  first i thought it was my psu, i changed it but still the same issue. 

[peach-gomo-6.json](https://github.com/jnthas/clockwise/files/12887255/peach-gomo-6.json)

[peach-gomo-22.json](https://github.com/jnthas/clockwise/files/12887261/peach-gomo-22.json)

jnthas, 2023-10-13:
Rob, some considerations
1) Color: I noticed the bg purple was different of image bg. To get the correct color I use the color picker of iamge editor to get the HEX notation from the image background and then I set the color code in the background like I did here
[image]
You are correct, there is no transparent color on Clockwise, so the backgroiund color of the image must be set during the edit.

2) Image size: The ESP32 have a limited memory, when I'm creating a clockface I try to reduce the image size as much as possible each byte matter here. For example, in the peach-gomo-22 the image size is 50x50 but the really important area, which have the animation is 50x36. When you crop the image you are saving 70 pixels of 16bit each for a single frame.

3) It seems the flickering was caused by the "datetime" element in json. The field "content" was empty but it is required since it defines the format of date/time.

4) One thing that it still doesn't work is, if you have the animation of 64x64 the date/time won't work because the clock is updating the animation and it replaces all elements in the display.

Here is the JSONs without flickering

[peach-gomo-6.json](https://github.com/jnthas/clockwise/files/12888777/peach-gomo-6.json)
[peach-gomo-22.json](https://github.com/jnthas/clockwise/files/12888778/peach-gomo-22.json)

robegamesios, 2023-10-13:
@jnthas thanks for the fix! I understand it better now 😀. 

1. Great on the bgColor tip👍.

2. True on the image size. I noticed anything more than 1kb will not work and only displays a blank screen on the panel. I found some sites that can edit gifs and export as png and also compress.

3. The first time I tried to remove the setup it didn't work for me hence I thought just removing the date time format will do, but I probably have a bad Json then.  

4. Yeah I need to leave space for the clock. Some themes I'll leave without, so it's more like an animated photo frame like Divoom.

I created a script to create a Json from a directory of images. Overall from converting the gif to creating the Json takes less than 5 min. I'm planning to create a video to show the process I figured to make themes. I'll share it when I create it, maybe tomorrow.

I forked your repo so I can make edits and quicker testing. I can also raise a PR to you if you like to include the themes I made so far so it becomes available on your site. Currently I'm testing using my local server per your wiki. 

This is a great project, kudos to you for making it!!

I also asked you on a separate issue I raised, more of a enhancement to have an audio visualizer. I just don't know how to go about it yet. Mainly how to map the panel with the sound. I installed an inmp441 mic to my board and can detect using the serial monitor, but that's all I have for now.

I'm thinking of building an ios app and have all the themes so that you can just select a theme and uploads to the panel, just an idea for now.

Thanks a bunch!

robegamesios, 2023-10-13:
@jnthas would be awesome if there is a way to separate the delay from the sprite animation, see video below. like set the delay for 1 minute, then the sprite will animate (jump) independently having its own timer and not rely on the global delay. then we can mimic the original mario clock. thoughts?

https://github.com/jnthas/clockwise/assets/10041871/7e8a015a-7bc9-42cd-9767-9188d94fb1e8

robegamesios, 2023-10-13:
hi @jnthas im having issue trying to upload the canvas clockface using platformIO. im getting this

[Time] NTP Server: time.google.com, Timezone: America/Los_Angeles
[HTTP] GET request to '192.168.50.10/snoopy3.json' on port 4443
[  4701][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():264]: (-32512) SSL - Memory allocation failed
[  4702][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -32512
Connection failed
deserializeJson() failed: EmptyInput

the other clock faces (mario, pacman) works fine on uploading via VSC. 

is the canvas repo updated? coz i copied the code there and pasted it in the clockwise firmware clockfaces local folder. when cloning the clockwise main repo, the cw-cf-0x06 and 0x07 folders are empty.

thanks

jnthas, 2023-10-14:
This error happens when you are not running the local server. It tries to get the clockface but the server didn't answer.
[HTTP] GET request to '192.168.50.10/snoopy3.json' on port 4443
Go to clock-club/local and run `python local-server.py`

jnthas, 2023-10-14:
> @jnthas would be awesome if there is a way to separate the delay from the sprite animation, see video below. like set the delay for 1 minute, then the sprite will animate (jump) independently having its own timer and not rely on the global delay. then we can mimic the original mario clock. thoughts?
> 
>  IMG_2678.mov

Totally agree. I have the same idea, but I would have to implement a dynamic array of timers and I removed from my MVP. But I agree that this solution would be better for sure. We can open an issue just for that.

jnthas, 2023-10-14:
I saw the issue about the audio visualizer, would be cool to have it. I think I have an inmp441 mic here but there are so many projects to manage I don't know I will have time for that haha.

robegamesios, 2023-10-14:
> This error happens when you are not running the local server. It tries to get the clockface but the server didn't answer. [HTTP] GET request to '192.168.50.10/snoopy3.json' on port 4443 Go to clock-club/local and run `python local-server.py`

hi @jnthas i do have the local server running. but the clockface cannot connect to it. im not seeing the logs in terminal.  is the canvas repo the same as the one in the clockwise master repo?

robegamesios, 2023-10-14:
> Totally agree. I have the same idea, but I would have to implement a dynamic array of timers and I removed from my MVP. But I agree that this solution would be better for sure. We can open an issue [...]

once i figure out the reason why running the canvas clockwise does not work when uploading from VSC, ill see if i can try to do this

robegamesios, 2023-10-14:
> I saw the issue about the audio visualizer, would be cool to have it. I think I have an inmp441 mic here but there are so many projects to manage I don't know I will have time for that haha.

yeah totally understandable :) thanks

jnthas, 2023-10-14:
> hi @jnthas i do have the local server running. but the clockface cannot connect to it. im not seeing the logs in terminal. is the canvas repo the same as the one in the clockwise master repo?

Yes, it works. I just ran it here locally using master branch. Check if the computer running the server and the clock are in the same network. Also, snoopy3 must be in the local/ folder

robegamesios, 2023-10-14:
See screenshots, i have the local server running. the canvas clock theme works if i upload from your website. but flashing using VSC has this issue.

is your WifiManager library version 2.0.16-rc.2?

im wondering if its the dependency libraries that's causing the issue?

[image]

[image]
[image]
[image]
[image]

jnthas, 2023-10-14:
Rob, observing your screenshots, it looks ok. Answering your question about WifiManager, it using the 2.0.15-rc.1 here. 
Analyzing with more attention now, I can see you are getting a different error I supposed to be. When the python server is not running you will see something like this: 
```
[WiFi] Connected to HUAWEI-5R42NS_HiLink, IP address 192.168.3.17
[Time] NTP Server: time.google.com, Timezone: America/Sao_Paulo
[HTTP] GET request to '192.168.3.19/peach-gomo-22.json' on port 4443
[  4203][E][ssl_client.cpp:129] start_ssl_client(): socket error on fd 49, errno: 104, "Connection reset by peer"
[  4204][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -1
Connection failed
deserializeJson() failed: EmptyInput 
```
In the end, the json is empty and the clockwise can't deserialize but when I'm not running the python webserver, I get an "Connection reset by peer" and you got a different one that's new to me "SSL - Memory allocation failed". :thinking: 

I will send all library versions I'm using here locally. Let's see if is something related to it. 

```
name=Adafruit BusIO
version=1.12.0
name=Adafruit GFX Library
version=1.11.3
name=ArduinoJson
version=6.21.2
name=ESP32 HUB75 LED MATRIX PANEL DMA Display
version=2.0.7
name=ezTime
version=0.8.3
name=Improv WiFi Library
version=0.0.1
name=PNGdec
version=1.0.1
name=WiFiManager
version=2.0.15-rc.1
Unity is unit tests, don't worry
```

jnthas, 2023-10-14:
Another thing, when you compile what's the use of memory in your case, this is mine

```
Dependency Graph
|-- ESP32 HUB75 LED MATRIX PANEL DMA Display @ 2.0.7+sha.38eaea7
|-- Adafruit GFX Library @ 1.11.3
|-- Adafruit BusIO @ 1.12.0
|-- SPI @ 2.0.0
|-- Wire @ 2.0.0
|-- ezTime @ 0.8.3
|-- Improv WiFi Library @ 0.0.1+sha.60915cb
|-- WiFiManager @ 2.0.15-rc.1+sha.aa8856a
|-- ArduinoJson @ 6.21.2
|-- PNGdec @ 1.0.1
|-- cw-commons
|-- canvas
Building in release mode
Retrieving maximum program size .pio/build/esp32dev/firmware.elf
Checking size .pio/build/esp32dev/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  29.9% (used 98104 bytes from 327680 bytes)
Flash: [========= ]  85.1% (used 1115689 bytes from 1310720 bytes)
```

jnthas, 2023-10-14:
If you search for "SSL - Memory allocation failed esp32" you will see many issues most of them related to out of memory. But I still didn't get why is happening with you and not with me (for a while).

robegamesios, 2023-10-14:
> Another thing, when you compile what's the use of memory in your case, this is mine
>
> ```
> Dependency Graph
> |-- ESP32 HUB75 LED MATRIX PANEL DMA Display @ 2.0.7+sha.38eaea7
> |-- Adafruit GFX Library @ [...]

here is mine

Dependency Graph
|-- ESP32 HUB75 LED MATRIX PANEL DMA Display @ 3.0.9+sha.3c7173a
|-- Adafruit GFX Library @ 1.11.9
|-- Adafruit BusIO @ 1.14.5
|-- SPI @ 2.0.0
|-- Wire @ 2.0.0
|-- ezTime @ 0.8.3
|-- Improv WiFi Library @ 0.0.2+sha.754d9f2
|-- WiFiManager @ 2.0.16-rc.2+sha.f328cd0
|-- ArduinoJson @ 6.21.3
|-- PNGdec @ 1.0.1
|-- cw-commons
|-- canvas

robegamesios, 2023-10-14:
here is mine next to yours

name=Adafruit BusIO
version=1.12.0 ------------------------->1.14.5
name=Adafruit GFX Library
version=1.11.3 ------------------------->1.11.9
name=ArduinoJson
version=6.21.2 ------------------------>6.21.3
name=ESP32 HUB75 LED MATRIX PANEL DMA Display
version=2.0.7 ------------------------->3.0.9+sha.3c7173a
name=ezTime
version=0.8.3 ------------------------->0.8.3
name=Improv WiFi Library
version=0.0.1 ------------------------->0.0.2+sha.754d9f2
name=PNGdec
version=1.0.1 ------------------------->1.0.1
name=WiFiManager
version=2.0.15-rc.1 ------------------->2.0.16-rc.2+sha.f328cd0

robegamesios, 2023-10-14:
i tried matching your version but i still get the same SSL - memory allocation error.  So not sure how to fix my issue :( .  If i use the web flash from your site, it works fine.

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
default_envs = esp32dev
name = Clockwise

[env:native]
platform = native
test_framework = unity
test_ignore = test_embedded

[env:esp32dev]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
test_ignore = test_native
monitor_speed = 115200
lib_deps = 
	https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA.git
	adafruit/Adafruit GFX Library@1.11.3
	adafruit/Adafruit BusIO@1.12.0
	SPI@2.0.0
	Wire@2.0.0
	ropg/ezTime@0.8.3
	https://github.com/jnthas/Improv-WiFi-Library#v0.0.1
	https://github.com/tzapu/WiFiManager#v2.0.15-rc.1
	bblanchon/ArduinoJson@6.21.2
	bitbank2/PNGdec@1.0.1
build_src_filter = +<*> -<.git/> -<.svn/> -<example/> -<examples/> -<test/> -<tests/> -<clockfaces/>
build_flags = 
	-D CW_FW_VERSION="\"1.4.1\""
	-D CW_FW_NAME="\"${sysenv.FW_NAME}\""

jnthas, 2023-10-15:
I don't know the cause but we can do remove the SSL part if you want to run locally only. 
I've created a branch called `insecure-wifi` with the changes on both repositories clockwise/insecure-wifi and cw-cf-0x07/insecure-wifi 
Please, switch to that branches, it will reduce in 10% the Flash size and I believe the RAM too.
```
RAM:   [===       ]  29.6% (used 97136 bytes from 327680 bytes)
Flash: [========  ]  75.7% (used 992213 bytes from 1310720 bytes)
```

Also, to serve the json you need to run simple Python server doing `$ python3 -m http.server 4443`

Let me know if it works

robegamesios, 2023-10-15:
> I don't know the cause but we can do remove the SSL part if you want to run locally only. I've created a branch called `insecure-wifi` with the changes on both repositories clockwise/insecure-wifi [...]

hey thanks ill be back in 2 hours and will try, how do i access the settings? what local address? thanks

jnthas, 2023-10-15:
Rob, get the local address in the logs when you start the clockwise. 
```
[WiFi] Connected to HUAWEI-5R42NS_HiLink, IP address 192.168.3.17      <---- here
[Time] NTP Server: time.google.com, Timezone: America/Sao_Paulo
[HTTP] GET request to '192.168.3.19/snoopy3.json' on port 4443
[  5175][E][WiFiClient.cpp:268] connect(): socket error on fd 49, errno: 104, "Connection reset by peer"
Connection failed
```

If you flash it locally, using platformio, you don't need to set the wifi because it is kept. Different form the web flash, when you flash using platformio, it just update the program part not the settings, so your previous wifi configuration is kept.

Basically what I did was, change the branches and flash to my clock. Start the simple server using that command I sent and that is it. I did change anything in the setting page. 

[image]

robegamesios, 2023-10-15:
@jnthas thanks! i had to reflash my esp32 to clear all memory. I still had the mem alloc issue using the master branch though. Good news is the wifi-insecure works!  thanks a lot!

ill see if can try to update the sprite interval separately

i also started looking into the OTA update. although i was able to make it work with the Mario branch that has INO file, im still not having success using the PlatformIO part. my initial idea is to just have a button in the settings that would launch the other page that will do the flashing.  ill keep you posted

jnthas, 2023-10-15:
Great, thank you! I will close the issue

closed 2023-10-15 by jnthas