jnthas/clockwise #54: how to set raw.githubusercontent.com to point to forked directory? (just a question)

robegamesios, 2023-10-18:
hi @jnthas can you show me how to reconfigure raw.githubusercontent.com to point to my forked shared folder? thank you

robegamesios, 2023-10-19:
if i enter an invalid (or unreachable) hostname, the panel gets stuck with the Splash screen "Downloading" screen. 

not sure why the panel cannot download the raw json (e.g. https://raw.githubusercontent.com/robegamesios/clock-club/main/shared/donkey-kong.json) and just gets stuck with the splash screen

i updated this:
```if (server.startsWith("raw.")) {
    port = 443;
    file = String("/robegamesios/clock-club/main/shared" + file);
  }```
is there another place i should update? thanks

robegamesios, 2023-10-20:
@jnthas i figured out the issue of the SSL- memory allocation failed when trying to build on PlatformIO.  you will need to downgrade the espressif32 framework

`platform = espressif32 @ ^5.4.0`

that fixed the issue. I am also able to access the clockfaces for canvas on my forked repo.

https://github.com/robegamesios/clockwise/blob/main/firmware/platformio.ini

just incase someone else encounters this issue. 

closing this

closed 2023-10-20 by robegamesios

jnthas, 2023-10-20:
Great. Thank you.