jnthas/clockwise #68: Matrix Chaning

alawSD, 2024-05-06:
Would it be possible to add matrix chaining to use 2 64x32 as one 64x64 as described in the ESP32 HUB75 MatrixPanel? 

https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA?tab=readme-ov-file#can-i-chain-panels

Currently chaining 2 panels repeats the image as two 64x32.

jnthas, 2024-05-07:
By default, not. We need to make some changes unfortunately.

krzimmer, 2025-04-12:
Hi @jnthas! Related question. Can the ESP32 handle chaining two 64x64 panels? I was looking around in the code for references to height, width and changed all from 64, 64 to 128, 64. That didn't do the trick. 
Could you explain the other limitations that would prevent 128x64?

jnthas, 2025-04-21:
Hello @krzimmer, Yes, it's possible to chain displays. Check the display library for more details
https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA?tab=readme-ov-file#further-information