jnthas/clockwise #69: error when I try to build pacman clockface

makerawei, 2024-05-07:
I put PNGdec in clockwise\components and add a CMakeLists.txt file as below  
```cmake
cmake_minimum_required(VERSION 3.15)

if(ESP_PLATFORM)
	# Build ArduinoJson as an ESP-IDF component
	idf_component_register(INCLUDE_DIRS src)
	return()
endif()

project(PNGdec VERSION 1.0.1)
add_subdirectory(src)

```

```bash
cmd.exe /C "cd . && D:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  @CMakeFiles\clockwise.elf.rsp -o clockwise.elf  && cd ."
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/cw-cf-0x07/libcw-cf-0x07.a(Clockface.cpp.obj):(.literal._ZL9openImagePKc+0x10): undefined reference to `PNG::openRAM(unsigned char*, int, void (*)(png_draw_tag*))'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/cw-cf-0x07/libcw-cf-0x07.a(Clockface.cpp.obj):(.literal._ZL18getImageDimensionsPKcRhS1_+0x0): undefined reference to `PNG::getWidth()'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/cw-cf-0x07/libcw-cf-0x07.a(Clockface.cpp.obj):(.literal._ZL18getImageDimensionsPKcRhS1_+0x4): undefined reference to `PNG::getHeight()'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/cw-cf-0x07/libcw-cf-0x07.a(Clockface.cpp.obj):(.literal._ZL18getImageDimensionsPKcRhS1_+0x8): undefined reference to `PNG::close()'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/cw-cf-0x07/libcw-cf-0x07.a(Clockface.cpp.obj):(.literal._ZL7PNGDrawP12png_draw_tag+0x0): undefined reference to `PNG::getLineAsRGB565(png_draw_tag*, unsigned short*, int, unsigned int)'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/cw-cf-0x07/libcw-cf-0x07.a(Clockface.cpp.obj):(.literal._ZL11renderImagePKchh+0x0): undefined reference to `PNG::decode(void*, int)'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/cw-cf-0x07/libcw-cf-0x07.a(Clockface.cpp.obj): in function `openImage(char const*)':
D:/github/clockwise/firmware/clockfaces/cw-cf-0x07/PNGRender.h:35: undefined reference to `PNG::openRAM(unsigned char*, int, void (*)(png_draw_tag*))'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/cw-cf-0x07/libcw-cf-0x07.a(Clockface.cpp.obj): in function `getImageDimensions(char const*, unsigned char&, unsigned char&)':
D:/github/clockwise/firmware/clockfaces/cw-cf-0x07/PNGRender.h:59: undefined reference to `PNG::getWidth()'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/github/clockwise/firmware/clockfaces/cw-cf-0x07/PNGRender.h:59: undefined reference to `PNG::getHeight()'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/github/clockwise/firmware/clockfaces/cw-cf-0x07/PNGRender.h:60: undefined reference to `PNG::close()'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/cw-cf-0x07/libcw-cf-0x07.a(Clockface.cpp.obj): in function `PNGDraw(png_draw_tag*)':
D:/github/clockwise/firmware/clockfaces/cw-cf-0x07/PNGRender.h:25: undefined reference to `PNG::getLineAsRGB565(png_draw_tag*, unsigned short*, int, unsigned int)'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/cw-cf-0x07/libcw-cf-0x07.a(Clockface.cpp.obj): in function `renderImage(char const*, unsigned char, unsigned char)':
D:/github/clockwise/firmware/clockfaces/cw-cf-0x07/PNGRender.h:49: undefined reference to `PNG::decode(void*, int)'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/github/clockwise/firmware/clockfaces/cw-cf-0x07/PNGRender.h:49: undefined reference to `PNG::close()'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/arduino/libarduino.a(WiFiClientSecure.cpp.obj):(.literal._ZN16WiFiClientSecure4stopEv+0x4): undefined reference to `stop_ssl_socket(sslclient_context*, char const*, char const*, char const*)'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/arduino/libarduino.a(WiFiClientSecure.cpp.obj):(.literal._ZN16WiFiClientSecure5writeEPKhj+0x0): undefined reference to `send_ssl_data(sslclient_context*, unsigned char const*, unsigned int)'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/arduino/libarduino.a(WiFiClientSecure.cpp.obj):(.literal._ZN16WiFiClientSecure4readEPhj+0x0): undefined reference to `get_ssl_receive(sslclient_context*, unsigned char*, int)'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/arduino/libarduino.a(WiFiClientSecure.cpp.obj):(.literal._ZN16WiFiClientSecure9availableEv+0x0): undefined reference to `data_to_read(sslclient_context*)'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/arduino/libarduino.a(WiFiClientSecure.cpp.obj):(.literal._ZN16WiFiClientSecureC2Ev+0x4): undefined reference to `ssl_init(sslclient_context*)'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/arduino/libarduino.a(WiFiClientSecure.cpp.obj):(.literal._ZN16WiFiClientSecure7connectEPKctS1_S1_S1_+0xc): undefined reference to `start_ssl_client(sslclient_context*, char const*, unsigned int, int, char const*, bool, char const*, char const*, char const*, char const*, bool, char const**)'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/arduino/libarduino.a(WiFiClientSecure.cpp.obj): in function `WiFiClientSecure::stop()':
D:/github/clockwise/components/arduino/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:98: undefined reference to `stop_ssl_socket(sslclient_context*, char const*, char const*, char const*)'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/arduino/libarduino.a(WiFiClientSecure.cpp.obj): in function `WiFiClientSecure::write(unsigned char const*, unsigned int)':
D:/github/clockwise/components/arduino/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:188: undefined reference to `send_ssl_data(sslclient_context*, unsigned char const*, unsigned int)'
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/arduino/libarduino.a(WiFiClientSecure.cpp.obj): in function `WiFiClientSecure::read(unsigned char*, unsigned int)':
D:/github/clockwise/components/arduino/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:218: undefined reference to `get_ssl_receive(sslclient_context*, unsigned char*, int)'
[... cut ...]
d:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/arduino/libarduino.a(WiFiClientSecure.cpp.obj): in function `WiFiClientSecure::connect(char const*, unsigned short, char const*, char const*)':
D:/github/clockwise/components/arduino/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:149: undefined reference to `start_ssl_client(sslclient_context*, char const*, unsigned int, int, char const*, bool, char const*, char const*, char const*, char const*, bool, char const**)'
```

jnthas, 2024-05-07:
Got it. Let me ask @abrender maybe he knows the issue.

makerawei, 2024-05-11:
@abrender could you please help me to solve the error；）thanks a lot !

jnthas, 2024-05-11:
@makerawei take a look at this branch, i was able to build it without errors -> https://github.com/jnthas/clockwise/pull/70

I dont know you are aware but you don't need to use esp-idf to compile the project, just open it in platform.io and it will work fine with arduino framework

makerawei, 2024-05-13:
@jnthas thanks a lot；）  
I tested your commit, and for success building there should be a CMakeList.txt in PNGdec  
```cmake_minimum_required(VERSION 3.15)

if(ESP_PLATFORM)
	idf_component_register(INCLUDE_DIRS src)
	return()
endif()

project(PNGdec VERSION 1.0.1)
add_subdirectory(src)
```
And a code line in  cw-cf-0x07/Clockface.cpp:198 is always true.
```C
bool shouldReturnToOrigin = doc["loop"][sprite->_spriteReference]["shouldReturnToOrigin"].as<bool>() ? : false;
```
this code would be error in my ESP-IDF env.  
After change it to   
```C
bool shouldReturnToOrigin = doc["loop"][sprite->_spriteReference]["shouldReturnToOrigin"].as<bool>() ? true : false;
```
I build pacman success !

jnthas, 2024-05-13:
Great! As PNGdec don't have the CMakesList file, you could open a PR there if you want.