jnthas/clockwise #71: Information please ?

spordiab, 2024-05-16:
hello first of all I wanted to thank you for your work.  I would like to understand if the poster is normal or there is a bug?  The PacMan clock turns red from time to time.  THANKS
[image]

jnthas, 2024-05-18:
Hi @spordiab! It's looks there is some misconnection. Please review your wiring again. Half of display is red, probably some section A,B,C,D,E in the display is wrong.

Kalki2898add, 2024-07-30:
Can i use 2 p5 32 x 64 display connect series. Will it work ?

jnthas, 2025-10-14:
No, only 64x64 ones are supported. I'm closing this issue, feel free to re-open if necessary