jnthas/clockwise #72: Timezone error for America/Santiago

joyarzun, 2024-05-28:
Hello there
I've setup all and it's working but the timezone still appears to be incorrect. I've set America/Santiago and it should show 21:50 but it's showing 1:50. I did change to America/Sao_Paulo and it works. I've changed the NTP server to others but it still continue to showing the same. What else can I do?

joyarzun, 2024-05-29:
More info:
When I've set the Posix timezone for America/Santiago that is "<-04>4<-03>,M9.1.6/24,M4.1.6/24" the browser does a request like this:
`http://192.168.0.xxx/set?manualPosix=%3C-04%3E4%3C-03%3E,M9.1.6/24,M4.1.6/24`
so it is doing a parsing of the symbols < and >. Then I did the request with cURL without doing the parsing and it worked:

`curl 'http://192.168.0.xxx/set?manualPosix=<-04>4<-03>,M9.1.6/24,M4.1.6/24' \
  -X 'POST' \
  -H 'Accept: */*' \
  -H 'Accept-Language: en-US,en;q=0.9,es-CL;q=0.8,es;q=0.7,pt;q=0.6' \
  -H 'Connection: keep-alive' \
  -H 'Content-Length: 0' \
  -H 'Origin: http://192.168.0.209' \
  -H 'Referer: http://192.168.0.209/' \
  -H 'User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/125.0.0.0 Safari/537.36' \
  -H 'dnt: 1' \
  -H 'sec-gpc: 1' \
  --insecure`

So the UI is failing to support < and > symbols that is Posix standard for some timezones

jnthas, 2024-05-29:
Hi @joyarzun! It seems a bug in the manualPosix, maybe we should force avoiding URL encode. I'm wondering why it didn't work when you set America/Santiago. Have you tried set other NTP server that is not Google? It should work with timezone name.

joyarzun, 2024-06-02:
> Hi @joyarzun! It seems a bug in the manualPosix, maybe we should force avoiding URL encode. I'm wondering why it didn't work when you set America/Santiago. Have you tried set other NTP server that is not Google? It should work with timezone name.

Indeed, I've tried with others ntp servers but the result is the same, also I've tried with the eztime lib without success but it's works with America/Sao_Paulo or Pacific/Auckland, so I think we have two issues, first is a bug the the lib (this repo is affected only) and the second is the parsing of Posix timezone

x0SiN0x, 2025-02-19:
Has there been any progress on this?  I loaded the latest version using America/Denver and its not correct

DocGEEK27, 2025-02-20:
Hey. I have a same problem with Europe :/

x0SiN0x, 2025-02-20:
Finally got mine working using the "Posix Timezone String".  I left the timezone alone and just added "MST7MDT,M3.2.0,M11.1.0" to that Posix setting and working great now.  Actually boots faster too which was an interesting side effect.

jnthas, 2025-02-22:
There's something else with those timezones that I can't replicate here.. I mean, the timezone works fine for me (yes, it works on my machine :D) I just set the timezone to Denver, and it displays correctly. 

[image]

[image]

jnthas, 2025-02-22:
Also, please check this issue, it can be the reason 
https://github.com/jnthas/clockwise/issues/39

x0SiN0x, 2025-02-22:
Ahhh, could be that part mentioned about "port 2342 (UDP)", I allow out NTP (standard 123) and various other required traffic but port 2342 is not on that list.  I will need to test this once i get back but i bet that is the case.

closed 2025-10-14 by jnthas