jnthas/clockwise #73: Panel (FM6126) DCYH-P2-6464-1515-VP seems to overlay second and fourth section with first and third

trantoriana, 2024-07-20:
Hi,

I have following panel:

* DCHY-P2-6464-1515-VP

with layout:

* https://github.com/esden/led-panel-docs/tree/master/DCHY-P2-6464-1515-VP

Using a NODEMCU ESP32 Dev KIT connecting following pins on the HUB 75 E:

* R1: 25
* B1: 27
* R2: 14
* B2: 13
* A: 23
* C: 5
* CLK: 16
* OE: 15

* G1: 26
* GND
* G2: 12
* E: 32
* B: 19
* D: 17
* LAT: 4
* GND

After flashing the display and the clock works but the display seems to not order the 4 section properly and overdraws them.
(Will try to attach a picture of this).
Seems positioning is wrong or the panel runs in 32 row mode?

Any pointers to what might be the issue?

Thanks, Trant

trantoriana, 2024-07-20:
[image]
[image]

Do not mind the actual wifi connection error. That works, but the picture shows nicely that the top part of the wifi symbol is positioned in the first segment and not the second.

Hope this helps.

Thanks, Trant

jnthas, 2024-07-20:
Hi Trant! It's possibly a wiring problem, is it possible to follow this scheme here ->  
https://github.com/jnthas/clockwise/blob/gh-pages/static/images/display_esp32_wiring_bb.png

richms, 2024-09-28:
The board I got off aliexpress does this too. They have their own github which is somewhat annoying if I cant use the default web flasher. https://github.com/sjh007/hub75-64-64 - would be good if this was able to be compiled for that board too without me having to install a whole development environment.

wizardelec, 2025-07-09:
Did anyone ever solve this?
I have a Chinese board that does this when flashed through clockwise - when I flash using the AliExpress vendors GitHub on platfformio the screen is as it should be but green and blue is swapped - the correction toggle in settings does nothing!
Getting really frustrated with it!!

jnthas, 2025-07-10:
Hi @wizardelec. Sorry to hear that, unfortunately those boards don't have relation with Clockwise. Since it's a open source project, I can't guarantee compatibility with all boards. I'm working on a custom and "official" board for Clockwise. I recommend you to raise an issue in the Aliexpress requesting a solution for that.

closed 2025-10-14 by jnthas