jnthas/clockwise #76: Thanks ^^

Hypogluco, 2024-11-22:
Hi, I am in the process of making personalized modifications to display the blood sugar level and trending arrows with customization of colors with immediate change from the web page. I really want to add ghosts from the Pacman game ^^, there are a lot of files to check. Short demonstration video

https://github.com/user-attachments/assets/63c39e38-405e-4243-9ce8-94240736d542

DocGEEK27, 2024-11-22:
Hi. I hope you manage to add the ghosts, it's going to be great. Thank you
for your nice job

JeffWDH, 2024-11-22:
Pacman is infuriating when it comes down to the last couple pellets... It never seems to go the right direction lol

I've attempted a couple times to make it smarter but I usually just make it dumber

jnthas, 2024-11-23:
That's really awsome! Looking forward to this!

Hypogluco, 2024-11-24:
Hi all, a little feedback on my progress, I managed to add a ghost ^^, I'm continuing the project.
To make Pacman smarter that would be great too, I'll look at that later ^^
Thanks

Uploading 20241124-184710-lmc-84_ngRyyrAw.mp4…

DocGEEK27, 2024-11-24:
Hi, thanks for your feedback. I can't wait to get the ghosts. Your clock is
awesome

Hypogluco, 2024-11-27:
Everything is good, the ghosts, the change of map colors during reset.

https://github.com/user-attachments/assets/bef34c6b-9192-4696-8723-e290ea54cbc9

DocGEEK27, 2024-11-27:
Nice work, it's a script or random ?

Hypogluco, 2024-11-27:
> Nice work, it's a script or random ?

For the pacman demo?

DocGEEK27, 2024-11-27:
Yes for the ghost

Hypogluco, 2024-11-27:
the same process as pacman, random

Hypogluco, 2024-11-27:
as soon as it is finished, I will place the code on the ghitub ^^

jnthas, 2024-12-05:
Open a PR for that!

Hypogluco, 2024-12-06:
I created a version without glycemia but with the weather temperature with Api Openweather. I don't know how ghitub works, I'm creating the page and then I'll post the files.

The code is in French, I'm going to a second version for the web page in English

https://github.com/user-attachments/assets/71a14fdd-7a41-4c81-b677-bb3cf9722bc3

vjeko2404, 2025-01-05:
Thats just funtastic and something the pacman clock really needs! 
I admire your progress!

will you post the link on your page when youre done?

closed 2025-10-14 by jnthas