jnthas/clockwise #79: timezone problem

rookie1988, 2024-12-31:
i have built my own clockwise. i am in China, but when I changed the timezone to   Asia/Shanghai or Asia/Hong_Kong and the NTPserver to those of Chinese companies, it is still not working, always 8 hours late. Can anyone please tell me why.

jnthas, 2024-12-31:
Hi @rookie1988. That's too bad, sorry for that! Can you try to change the NTP time server, maybe it can fix the problem
https://gist.github.com/mutin-sa/eea1c396b1e610a2da1e5550d94b0453. Or try to find a NTP server from China maybe. 

If it still doesn't work, try to use the POSIX timezone, I know some users could fix the problem with this 
https://github.com/ropg/ezTime#timezones-1

Both changes are done in the config web page of clockwise.

rookie1988, 2025-01-01:
> Hi @rookie1988. That's too bad, sorry for that! Can you try to change the NTP time server, maybe it can fix the problem https://gist.github.com/mutin-sa/eea1c396b1e610a2da1e5550d94b0453. Or try to [...]

Thank you so much for answering my question. I will try the options that you provided to see if it works. Such a great project that you have made. Thanks again.

Chenpeel, 2025-03-29:
[image]

set-up like this

Result is as following:

[image]

jnthas, 2025-03-30:
Hi @Chenpeel , are you reporting a problem in timezone or the display colors?

Chenpeel, 2025-03-31:
> Hi @Chenpeel , are you reporting a problem in timezone or the display colors?

thanks for your reply，I'm replying him for the Asia/Shanghai time problem. And there's a phenomenon that use cn NTP server, must restart three or more times, the board finally shows right time. (NTP: cn.pool.ntp.org)