jnthas/clockwise #80: The Time on Pacman Clockface

bizleung, 2025-01-07:
Hello Jnthas,

Thank you so much for your hard work on this project. I really like the Mario Bros and Pacman clockfaces. I recently switched the clockface from Mario Bros to Pacman, and I absolutely love it. However, I noticed a problem: the time on the Pacman clockface seems to be inaccurate, sometimes running several seconds or even tens of seconds slow. On the other hand, the time on the Mario Bros clockface is very accurate. The only change I made was switching the clockface, and all other settings remain the same. I just wanted to let you know about this issue.

Thanks again!

jnthas, 2025-01-14:
Hey, @bizleung. This is new to me. I've used the pacman for a long time in my office and it was never slow. 🤔 The library checks the NTP server every time, so it's should not happen.

bizleung, 2025-01-15:
So yeah, I also find it pretty weird. The settings are definitely fine, but the time is still not accurate. What's even stranger is that after every reboot, the delay is different—sometimes it's just a few seconds, and other times it’s almost a minute. I’ve tried several NTP servers too, but it’s the same issue.

https://github.com/user-attachments/assets/f895face-2502-4972-a683-69a882d62b16

jnthas, 2025-10-14:
Hey @bizleung, I'm going to close this issue due to inactivity. Feel free to reopen it if necessary

closed 2025-10-14 by jnthas

Meekdai, 2025-10-17:
I also encountered the same problem, it was indeed a few seconds slower.
@bizleung Have you found the reason?