jnthas/clockwise #81: how to set the parameters?

ericoud, 2025-01-08:
Hi, Great project, I'd like to build the clock but I'm running into a problem I can't solve.
I am not an expert... I used PlatformIO to compile the program on an ESP32-S3. the compile and upload is OK, but when the program runs, I have the following messages in the serial monitor window:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377b8d
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2920
entry 0x403c98d8
[  1194][E][Preferences.cpp:483] getString(): nvs_get_str len fail: timeZone NOT_FOUND
[  1194][E][Preferences.cpp:483] getString(): nvs_get_str len fail: wifiSsid NOT_FOUND
[  1198][E][Preferences.cpp:483] getString(): nvs_get_str len fail: wifiPwd NOT_FOUND
[  1206][E][Preferences.cpp:483] getString(): nvs_get_str len fail: ntpServer NOT_FOUND
[  1213][E][Preferences.cpp:483] getString(): nvs_get_str len fail: canvasFile NOT_FOUND
[  1221][E][Preferences.cpp:483] getString(): nvs_get_str len fail: canvasServer NOT_FOUND
[  1229][E][Preferences.cpp:483] getString(): nvs_get_str len fail: manualPosix NOT_FOUND

I configured the ssid and wifi password in the load() section of the CWPreferences.h file as follows:
wifiSsid = preferences.getString(PREF_WIFI_SSID, "MySSID");
wifiPwd = preferences.getString(PREF_WIFI_PASSWORD, "MyPassword");

the ESP32 reboots indefinitely (watch dog: TG1WDT_SYS_RST)...
would you have an idea?
Thank you.

ericoud, 2025-01-09:
Build and download results:

Processing esp32dev (platform: espressif32@5.3.0; board: lolin_s3; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/lolin_s3.html
PLATFORM: Espressif 32 (5.3.0) > WEMOS LOLIN S3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6) 
 - tool-esptoolpy @ 1.40400.0 (4.4.0) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 44 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Adafruit GFX Library @ 1.11.11
|-- Adafruit BusIO @ 1.17.0
|-- SPI @ 2.0.0
|-- Wire @ 2.0.0
|-- ezTime @ 0.8.3
|-- Improv WiFi Library @ 0.0.2+sha.f02cb67
|-- WiFiManager @ 2.0.17+sha.8e4c612
|-- ArduinoJson @ 6.21.5
|-- PNGdec @ 1.0.3
|-- ESP32 HUB75 LED MATRIX PANEL DMA Display @ 3.0.11
|-- cw-commons
|-- timeinwords
Building in release mode
Retrieving maximum program size .pio/build/esp32dev/firmware.elf
Checking size .pio/build/esp32dev/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  14.7% (used 48036 bytes from 327680 bytes)
Flash: [=         ]  13.9% (used 911893 bytes from 6553600 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/cu.usbmodem14101
Forcing reset using 1200bps open/close on port /dev/cu.usbmodem14101
Waiting for the new upload port...
Uploading .pio/build/esp32dev/firmware.bin
esptool.py v4.4
Serial port /dev/cu.usbmodem14101
Connecting...
Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 98:3d:ae:e7:4b:04
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000eefff...
Compressed 14768 bytes to 10148...
Writing at 0x00000000... (100 %)
Wrote 14768 bytes (10148 compressed) at 0x00000000 in 0.2 seconds (effective 493.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 437.4 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 632.7 kbit/s)...
Hash of data verified.
Compressed 912288 bytes to 575784...
Writing at 0x00010000... (2 %)
Writing at 0x0001951f... (5 %)
Writing at 0x00023940... (8 %)
Writing at 0x0002e6c4... (11 %)
Writing at 0x000396fc... (13 %)
Writing at 0x000442db... (16 %)
Writing at 0x00049df2... (19 %)
Writing at 0x000501d2... (22 %)
Writing at 0x000570ef... (25 %)
Writing at 0x0005cd79... (27 %)
Writing at 0x000626cc... (30 %)
Writing at 0x00067e16... (33 %)
Writing at 0x0006ce31... (36 %)
Writing at 0x00071e2f... (38 %)
Writing at 0x00076d56... (41 %)
Writing at 0x0007bbf6... (44 %)
Writing at 0x00080b75... (47 %)
Writing at 0x00085bf7... (50 %)
Writing at 0x0008acf8... (52 %)
Writing at 0x0008fcf2... (55 %)
Writing at 0x00095653... (58 %)
Writing at 0x0009ac78... (61 %)
Writing at 0x0009fab0... (63 %)
Writing at 0x000a4bdf... (66 %)
Writing at 0x000a9b6c... (69 %)
Writing at 0x000aef35... (72 %)
Writing at 0x000b4483... (75 %)
Writing at 0x000ba2fa... (77 %)
Writing at 0x000bf821... (80 %)
Writing at 0x000c5316... (83 %)
Writing at 0x000cdc68... (86 %)
Writing at 0x000d5ef3... (88 %)
Writing at 0x000dcaa6... (91 %)
Writing at 0x000e280c... (94 %)
Writing at 0x000e7f9f... (97 %)
Writing at 0x000edddb... (100 %)
Wrote 912288 bytes (575784 compressed) at 0x00010000 in 4.6 seconds (effective 1596.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
--- Terminal on /dev/cu.usbmodem14101 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
=================================================== [SUCCESS] Took 65.85 seconds ===================================================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       SUCCESS   00:01:05.855
=================================================== 1 succeeded in 00:01:05.855 ===================================================

jnthas, 2025-01-14:
Hello @ericoud. If I not mistaken the message you pasted related with Preferences.cpp is just a warning. It's saying that there are no values stored since you just flash the clockwise. So, I believe this is not the cause of error. 

You mentioned that you are using ESP32-S3, this is not the ESP32 I'm use. So, first of all, can you confirm that it works when you flash your S3 via webflash, using the clockwise.page? 
Second, there are many causes for the error you mentioned "watch dog: TG1WDT_SYS_RST". What kind of power supply are you using?

ericoud, 2025-01-16:
Hello @jnthas, Thank you for your response.
My ESP32 was powered by my Mac USB port, without anything connected to it. It was just to check that the Wifi and web server worked. I used other programs that worked fine connected like this. For the web flash  it doesn't work, it told me that the ESP32-S3 was not supported...

[image]

I have another question, I have a 64 x64 panel made of 2 64 x 32 panel, I don't think I can use the web flash for this panel?
Same for the driver if they differ from the one you use, do the web server allows the modification of these parameters?

ericoud, 2025-01-16:
I modified the code on platformIO, to switch to virtual matrix using my 2-panel 64 x 32 matrix, and used a simple ESP32. It works great! 
thank you.

closed 2025-10-14 by jnthas